JamesP6000/WsprryPi #5: VNC vs WSPR

adaniel81, 2017-03-05:
Hello,
since I updated VNC 6.0.2,
after WPRS transmission, VNC server logs off and is no longer possible to connect remotely, VNC server goes wrong asking an account, but even with that you must restart the raspberry pi 2

JamesP6000, 2017-03-05:
That's interesting. WsprryPi bypasses the OS in many ways and directly accesses RAM, DMA engines, clock generators, etc. I'm guessing (i.e. I don't know) that VNC might be accessing the DMA engines also. I know that this isn't something that I'll be able to dig into though.

Do you absolutely need to control the Pi via VNC? You can also ssh into the Pi and launch WSPR that way. That is how I work on this code.

JamesP6000, 2017-04-17:
If you're using VNC then you might be using the audio driver which will conflict with WsprryPi. Try disabling audio?

JP

******
PWM Peripheral:
******
  The code uses the RPi PWM peripheral to time the frequency transitions
  of the output clock. This peripheral is also used by the RPi sound system
  and hence any sound events that occur during a WSPR transmission will
  interfere with WSPR transmissions. Sound can be permanently disabled
  by editing /etc/modules and commenting out the snd-bcm2835 device.

closed 2017-06-25 by JamesP6000