JamesP6000/WsprryPi #7: Double frequency.

alfille, 2017-04-08:


JamesP6000, 2017-04-10:
That's very odd... Can you send me the actual command line you used?

James

alfille, 2017-04-11:
Originally:

ExecStart=wspr -r -s -o  K1PHA FN42 20 20m

(part of a systemd script)

But then to test

wspr -t 14097100

gave a doubled rate on pin 4 (measured with a HP 5432A frequency
counter to the nearest Hz)

and now I use

# Seems to double the frequency
# so use 20 meter 14097100 / 2 = 7048550
ExecStart=/usr/bin/wspr -r -s -o  K1PHA FN42 20 7048550 0 0 0 0

Sucessfully (I added to extra pauses so as not to clutter up the airwaves)

JamesP6000, 2017-04-17:
I'm still not seeing this behavior... Can someone else confirm this? Are people receiving your transmissions on the double frequency?

JP

alfille, 2017-04-17:
This is a Pi 1B. And yes, my beacon is seen in Belgium Venezuela Texas (from Boston). (This is set at 7MH read at 14MHz=20m).

It's a standard Raspbian setup, headless.

closed 2017-04-17 by alfille