JamesP6000/WsprryPi #15: Wrong frequency output

Riccardo451, 2017-08-27:
good morning,
after installing a fresh jessie lite image on a raspberry pi zero w and running:
apt-get update, upgrade and dist-upgrade and installing WsprryPi I've run the following command:
sudo wspr -s -n mycallsign mylocator 10 20m.
By checking with the scope I see a square wave with a frequency that is the double of the requested frequency! I've tried also to run:
sudo wspr -s -n mycallsign mylocator 10 7100000
but I get a 14.200 Mhz output.
How I can fix this?

Riccardo451, 2017-08-28:
Redoing all the installation on a fresh image, I get after running sudo make:
cc -Wall -c mailbox.c
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 mailbox.o wspr.cpp -owspr
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 gpioclk.cpp -ogpioclk

k5dru, 2017-10-15:
I have this problem on "Raspberry Pi 3 Model B V1.2", but had to divide the frequency by 2.2 to get the correct frequency output on my scope. Even then I didn't get usable WSPR and the frequency was unstable. Screenshot of RX window on left and TX window on right: https://i.imgur.com/ixhubc5.jpg