JamesP6000/WsprryPi #18: Nothing happens but WiFi disconnection

Jibeji, 2018-03-12:
Hello,
I tried to install WsprryPi on a Raspberry Pi Zero W.
The --test-tone option works well but after running "wspr --repeat --offset --self-calibration MyCall MyLocator 20 30m", nothing happens but the WiFi link is disconnected, and of course no way to see the logs.

K0PIR, 2018-03-12:
Hello,

I haven’t used mine in a long time, but I believe it may be transmitting for you. When the WIFI disconnected on mine it was due to the transmission. Can you hook up a monitor to your Pi?

I never found a cure for that disconnect other thank hooking up a monitor or small LCD display. You could try the Pi forum here:https://groups.io/g/RaspberryPi-4-HamRadio <https://groups.io/g/RaspberryPi-4-HamRadio>

Rich, K0PIR
K0PIR.us

Jibeji, 2018-03-13:
Hi,
Solved by setting
`consoleblank=0`
in /boot/cmdline.txt

closed 2018-03-13 by Jibeji