JamesP6000/WsprryPi #19: Crashing when not launched manually

Jibeji, 2018-03-16:
Hi,
I would like WsprrPi to be launched at boot.

I tried to add the following sequence in /etc/rc.local, it crashes
`/usr/bin/sudo /usr/local/bin/wspr --repeat --offset --self-calibration MyCall MyLoc 20 30m 0`

So I tried to set it in the crontab without the --repeat option, it crashes.

By crashing I mean that not only there is no transmission, but it also make the WiFi fail.

Launching both command manually work fine.

dl9sec, 2018-04-13:
Hi,
i use the method for lauch at boot from the presentation here:
https://www.ham-yota.com/wp-content/uploads/2016/06/WSPR_YOTA2016.pdf
Installing this systemd sevice as described works fine for me:
https://github.com/oe3cts/WSPR-HAT/blob/master/Documents/wsprrypi.service

Regards, Thorsten
(BTW: I use the TX blinking LED modification and have a visual feedback if  wspr starts and transmitts)

Jibeji, 2018-04-13:
Thanks a lot for this very useful information.
73