JamesP6000/WsprryPi #26: WsprryPi on Pi 3b+

bcorky, 2019-01-31:
Has anyone got WsprryPi to work on the 3B+
With mine the software says” Pi version one detected” , runs once and hangs

AA7US, 2019-02-26:
bcorky, I'm successfully running WsprryPi on a 3B+ with Raspbian Stretch Lite.  No special problems noted other than the previously reported system lockups when attempting to run WsprryPi using a terminal inside the Pi's LXDE GUI desktop.  I disabled the on-board sound card and disabled the Pi's screen blanking (even though I run headless) and this seems to have solved the system lockup problem.  Everything works perfectly.

bcorky, 2019-02-26:
When it runs what version of the Pi does it say it detected?

AA7US, 2019-02-26:
Yes... "Detected Raspberry Pi version 2/3"

bcorky, 2019-02-26:
Interesting, I’ll make a new image and try again