JamesP6000/WsprryPi #29: Raspberry 3B+ Raspbian Buster -> Compiler Error: undefined reference to 'makedev'

hubertmeier, 2019-06-24:
Hi,

today i installed the brand new raspbian buster and i could not install WsprryPi becaus e make ends with the error message: **undefined reference to 'makedev'**

Thank you

berrak, 2019-06-30:
I can confirm above problem with gcc version 8 on Buster (Raspberry Pi v1).
gcc have had some major upgrades from Jesse (v. 4.9.2), Stretch (v. 6.3) to eight now in Buster thus I guess code is pretty much untouched for several years. My workaround was to install an old Jessie-image and make the binary this way. Compiles without any warnings/errors.

pi@wspr:~/GIT/WsprryPi $ make
cc -Wall -c mailbox.c
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 mailbox.o wspr.cpp -owspr
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 gpioclk.cpp -ogpioclk
pi@wspr:~/GIT/WsprryPi $ sudo make install
install -m 0755 wspr /usr/local/bin
install -m 0755 gpioclk /usr/local/bin

gibmat, 2019-07-21:
It's a pretty easy fix, just missing a header. I've got a commit in my fork: https://github.com/gibmat/WsprryPi/commit/bd8545954b9c43e704fe8178d93e98a5fef6b881

hubertmeier, 2019-07-23:
Yeah, it works!

Thank you very much 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

boxsection, 2019-10-14:
Works for me also. Just recompiled with gibmat's fork using buster on Pi 4.

nrnrp, 2019-10-19:
Hello, I have the same problem to install wsprryPi on Pi Zero W and raspbian buster.
Will it run when I do a new installation of an old jessie release?

cirrussecure, 2020-01-04:
Many thanks Mathais (gibmat).

I successfully applied your diff (https://github.com/gibmat/WsprryPi/commit/bd85459) and built wsprry!

cat /sys/firmware/devicetree/base/model -> 'Raspberry Pi 3 Model B Rev 1.2'
cat /etc/os-release -> 'PRETTY_NAME="Raspbian GNU/Linux 10 (buster) ...'

Best,
Tom KE2KQ

theventurousco, 2020-01-06:
Thanks @gibmat ! This worked!

gibmat, 2021-01-11:
Seeing as @JamesP6000 hasn't had any activity in this repo in almost four years, I'd consider it dead. It does look like @threeme3's repo (from which this one was originally forked) is a bit more active -- for one thing, a commit mentions support for the RPi4. There's also the explosion of forks, although I doubt many of them contain improvements.