JamesP6000/WsprryPi #30: Raspberry Pi Zero W Buster - Screen Lock or "Freeze" when try to transmit WSPR

SyCoTechs, 2019-08-27:
Im using the Raspberry Zero W with the newest 10.0 "Buster" Raspbian Build. Everything up to date.
I had first some problems in compiling but modify the files like that https://github.com/gibmat/WsprryPi/commit/bd8545954b9c43e704fe8178d93e98a5fef6b881 solved that problem.
After installed, transmitting and receiving of a test signal went quite well, done that multiple times on and off and no problems. 

The Problems started as i tried to transmit an actual WSPR Baken. As others i also experience the screen freeze problem. But it looks like i can come around with an additional info.
As others my screen also freezes when im connected over vnc and then start the transmitting.
It freezes at "Waiting for next WSPR Transmission window..."
But now the clue, when im connected over HDMI to my Computer screen, everything works fine!
Also the same time using vnc works wonderfull. 
The screen freezes as soon as i unplug the HDMI Cable and try to use vnc.
Even when a run with limited transmissions and then stop transmissions(--terminate 5) should be complete, the screen dont come to unfrozen again.

Plug in and out the HDMI Cable makes no difference at that point the screen shows frozen, also unplugging the router to maybe force vnc to stop doesnt shows an effect. Need to reboot to get it to work again.
I tried it headless direct into the CLI from boot with the same experience.

From my experiences i can tell that the device keeps sending in pulses after the screen got frozen, but it doesnt uses the usual break at 1:51 anymore. Wich leads me into that something is really broken as it seems like the system is freezing into loop or something. Additional the ZeroW dont accepts keyboard input anymore, it doesnt even serve it energy to light up his Capslock LED.

So it looks like it has something to do with screen driver or idk. I didnt experienced that problem at any other points, only with that software. FM Transmitting works nice and also the WSPR test-tone worked flawless.
When i let him connected over HDMI i can start and stop WSPR signal at any time over vnc and it works nicely.

My ZeroW is shown as Raspberry 1 btw, idk if this is wanted to be like this.
bc2836 modul is unloaded over blocklist.

I dont want to install an old firmware for the raspi, would be better if it could be get to running on the new Raspbian Buster. If all doenst helps i would probably do so but i hope theres a other solution somewhere.

CX1GU, 2019-09-08:
Hi SiCoTechs, i'm Gonzalo. Are you tried -f option? In my Zero w freeze problems come from NTP Frequency calc. I'm work from ssh and connected to my cellphone.

So, give a try to -f option and tell us.  CX1GU

SyCoTechs, 2019-09-10:
Hi Gonzalo,
yes i tried -f option with a specific frequency PPM correction with -p function which i calculated before, according to my SDR Stick receivings(SDR Stick is ppm corrected over GSM900 Tower).
That was one of my first tries.

Maybe my ZeroW is broken?
I dont know. But what i know is that with all other software and kinds of use the ZeroW works without problems. Also the FM transmitting with PiFM worked without any problems over VNC.
I only see these problems with WsprryPi

Regards
Dean

mark-orion, 2019-09-27:
I am having the same problem with Buster.
Usually I run wspr with the script below. It shuts down unused services (just in case...), syncs the time with ntpdate and triggers wspr in a while loop.
With this script I notice that the moment wspr starts transmitting the network goes down and comes back up after the 2min. wspr transmission.
I tried turning off WiFi power management as this is a well known culprit for many WiFi problems, but not here.

```
#!/bin/bash
# Run WSPR
CALL=yourcall
GRID=yourlocator
PWR=10
FRQ=20m
MAXWAIT=120
sudo systemctl stop vncserver-x11-serviced.service
sudo systemctl stop bluetooth.service
sudo systemctl stop systemd-timesyncd
sudo systemctl stop alsa-restore.service
sudo systemctl stop alsa-state.service
sudo systemctl stop triggerhappy.service
sleep 10
sudo systemctl daemon-reload
while :; do
    echo "Synchronizing Time"
    sudo ntpdate -b -p8 pool.ntp.org
    sudo wspr --terminate 1 --offset --self-calibration $CALL $GRID $PWR $FRQ
    NUMBER=$RANDOM
    let "NUMBER %= $MAXWAIT"
    echo "Waiting $NUMBER seconds..."
    sleep $NUMBER
done

```

normanr, 2020-06-19:
See also https://github.com/raspberrypi/linux/issues/3671, this seems to be affecting lots of people.

cathalferris, 2021-12-16:
To add another data point:

I've got a PiZero W version 1, with the QRPGuys [EZ-PiWspr](https://qrpguys.com/ha7dcd-ez-pi-wspr) board. 
I can transmit test tones perfectly fine, and I can easily pick these transmissions up with my SDR.

However, the Pi just hangs when trying to transmit WSPR signals. 

I've fully shielded the ez-pi board with Kaptan tape and copper foil, so I doubt there's any RF getting into the PiZero outside of the GPIO pins or the power cable.

When looking on the panadapter output on the HF transceiver, I often see a short burst of radio transmisison of about a symbol length and then silence. It's as though the Pi can't handle the first change in frequency.

In my trials of running the WSPR script via cronjob, I did get one complete WSPR transmission and then a hang of some type. I may yet just repurpose one of my earlier Pi systems to this task, as I suspect the PiZero just isn't up to the job.

zinkwazi, 2022-06-27:
Hi all, I posted re: the Raspberry Pi Zero 2 W crashin in https://github.com/JamesP6000/WsprryPi/issues/6 if anyone has made progress on this, please chime in.

thanks!

Greg

BobbyB256, 2022-07-15:
I have a fix! My Pi Zero W has been running Buster with a simple change to the wspr.cpp file. Around line 700 is the wait_every() function that waits for the one second after the even minute. In there are some calls to usleep() that compiles to something that locks the whole system! There might be other solutions but I have commented out those calls and just used the existing for loop and added another to look at the system time. The full function then looks like:
`// Wait for the system clock's minute to reach one second past 'minute'
void wait_every(
  int minute
) {
  time_t t;
  struct tm* ptm;
  for(;;){
    time(&t);
    ptm = gmtime(&t);
    if((ptm->tm_min % minute) == 0 && ptm->tm_sec == 0) break;
    //usleep(1000);
  }
  //usleep(1000000); // wait another second
  for(;;){
    time(&t);
    ptm = gmtime(&t);
    if(ptm->tm_sec == 1) break;
  }
}
`
Once you get this working you might find another Pi Zero specific problem - frequency drift. I was getting 4 Hz or more drift over the 2 minute transmission which means I was not getting reception reports. With an IR camera I measured 4 deg F temp rise evenly over the Broadcom chip during transmission, so the crystal sees about the same because it is small and right next to it. My solution here is to put heat sinks on top the Broadcom IC and bottom of the PC, and a small fan blowing over everything. Now I get drift of 0 or sometimes 1.

Happy WSPRing!

edegraaff, 2024-01-26:
All,  since i have my licence since a week or two (pd1eg) i wanted to try this project. 
After some reading and experimenting i was able to recieve the pi zero on my sdr, decode the signal and upload it to WSPRNet. 
It is correct the SSH sessions (wifi) is directly closed when you startup the program. I will do some experimentig. Thanks to the discussion here i have been able to transmit. Need to build the filter and so on. I start with some further testing and the first is adding a USB Nic to the zero and see if the connection stays.  And the first test shows adding an USB Nic to the PI the SSH connection stays ok. That means this could be a solution as temp fix. The main question that stays is there a Wifi disturbance during transmission thas is causing the "hang" of the system. I will try to do some extra investigation.

edegraaff, 2024-01-26:
Info: Pi Zero W , Raspbian Bullseye, SMA connector to GPIO4 and Ground 
Used the tip from the raspberry forum. 
I did not uncomment the code as sugested by Bobby 
Next step will be to connect the PI to the 20 meter loop.
Tested command  sudo ./wspr mycall mygrid 33 20m
The bandpass filter is not yet builded. 

_[Re: Need help with "undefined reference makedev"](https://forums.raspberrypi.com/viewtopic.php?t=250220#p1527764)
[Thu Aug 29, 2019 11:03 am](https://forums.raspberrypi.com/viewtopic.php?p=1527764#p1527764)

The problem is a missing function called makedev, not a program.
makedev() is actually a macro and you need to include the correct header for it. At the top of mailbox.c add the following line
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=250220#)

#include <sys/sysmacros.h>
I assume if the code used to compile under Stretch then one of the other includes used to include sysmacros but no longer does on Buster._

zinkwazi, 2024-01-31:
PD1EG, congrats on getting your license!
I will try the USB trick and see if that works.

(I did try recompile with BobbyB256's fix after it was posted but the issue persisted for me.)

edegraaff, 2024-02-01:
Thanks Greg,  Some more insights: 
USB keyboard not needed, i did transmissions with ony the monitor connected to dhe  hdmi connector. and tested the pizero. this is the result: i wil check more later.
[image]