JamesP6000/WsprryPi #31: I can confirm above problem with gcc version 8 on Buster (Raspberry Pi v1 B).

AdrianS-commits, 2019-10-24:
I can confirm above problem with gcc version 8 on Buster (Raspberry Pi v1 B).
gcc have had some major upgrades from Jessie (v. 4.9.2), Stretch (v. 6.3) to eight now in Buster thus I guess wspr-code is pretty much untouched for several years. My workaround was to install the latest released Jessie-image (2017-07-05) and make the binary this way. Compiles without any warnings/errors::

  pi@wspr:~/GIT/WsprryPi $ make
  cc -Wall -c mailbox.c
  g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 mailbox.o   wspr.cpp -owspr
  g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 gpioclk.cpp -ogpioclk
  pi@wspr:~/GIT/WsprryPi $ sudo make install
  install -m 0755 wspr /usr/local/bin
  install -m 0755 gpioclk /usr/local/bin

_Originally posted by @berrak in https://github.com/JamesP6000/WsprryPi/issues/29#issuecomment-507042667_

HELP! I am new to programming. - I just want to be able to use my Raspberry Pi 3B+ for WSPR and it seems IMPOSSIBLE - Can you please explain your step here - so I know exactly what to do - in the command line - please explain clearly as I really don't understand how you were able to get the latest version of Raspbian to work. Many thanks!

sp2ong, 2019-11-12:
Try to use this version:
https://github.com/g4wnc/WsprryPi

lryindra, 2019-11-12:
I also was successful of running WsprryPi on an up to date version of Raspbian: https://github.com/lryindra/WsprryPi

theventurousco, 2020-01-06:
I'm having the same problem.
Running an original Raspberry Pi (1st gen) 
uname data:
Linux raspberrypiI 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux

@AdrianS-commits can you tell me what you mean my install latest released jessie image?
I just installed NOOBS and then raspian (full version) on this Pi.
Should I start over with a different OS version?
Or can I do the necessary updates from the command line?

73 
Nick 
K6LJU

lryindra, 2020-01-07:
@theventurousco 
my uname data is 
          Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
and my gcc version is 
          gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
I was able to successfully build this version: 
          https://github.com/lryindra/WsprryPi
on my Rasberry Pi 3B

However, there is a step in the makefile that detects armv6.  I am unable to provide assistance because I do not have the older hardware.

diyit2, 2020-08-27:
I am having the same problem model B rev 2,
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="7"
> VERSION="7 (wheezy)"
> ID=raspbian
> ID_LIKE=debian
> ANSI_COLOR="1;31"

here's the failure message
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm
-DRPI1 mailbox.o wspr.cpp -owspr
cc1plus: error: unrecognized command line option ‘-std=c++11’
makefile:22: recipe for target 'wspr' failed
make: *** [wspr] Error 1
pi@raspberrypi ~/WsprryPi $

I have tried both the lyrinda and gw4nc

Any ideas?  Do I need an older version gcc to work with the this old wspr.cpp
tks

wodowiesel, 2020-08-27:
try -std=c11 or -std=iso9899:2011

diyit2, 2020-08-27:
thanks for responding
my apologies for ignorance on this end.  but
-std=c11 or -std=gnu11 or -std=iso9899:2011
do those statements replace statements in the makefile? 
or  are those alternative compilers that I should get and try?

I did notice in the makefile that they specifiy c++11
as below

**CXXFLAGS += -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5**

wodowiesel, 2020-08-27:
yes the c11 and iso specifications are the same, the gnu is c11 with gnu more extention functionality, just try the command without ++ its worth a try, still wondering why c++11 doesnt work, think it depends on the g++ version idk, i normally use only gcc for my stuff. maybe someone else can help

diyit2, 2020-08-28:
neither one of them worked.  The response is basicaly the same
original message
**cc1plus: error: unrecognized command line option ‘-std=c++11’**
modified makefile
**cc1plus: error: unrecognized command line option ‘-std=c11**
or
**cc1plus: error: unrecognized command line option ‘-std=iso9899:2011**

Its like it is not able to find the compiler or the invocation is not valid.

diyit2, 2020-08-28:
I wonder if it would be useful to try to build any kind of simple c code package that comes with the makefile etc just to see if the basic functions are working at all, never mind the wspr thing.  It seems like something fundamental is wrong somewhere.  I did install the latest raspbian ver 7.  I'm not savvy enough to go back a version.  its as if a path is not found or something like that.

diyit2, 2020-08-28:
it is not an overall problem, I just remembered that I built pi_vfo, 

https://github.com/JennyList/LanguageSpy/tree/master/RaspberryPi/rf/freq_pi

 although it did not consist of a makefile.  just simple gcc *

wodowiesel, 2020-08-28:
hmm if it doesn't recognize this parameters at all, then indeed something is basically wrong. did you install g++ or only gcc?

diyit2, 2020-08-28:
I don't know.  Do you know how I can tell if its there?
 I just grabbed the latest version for rpi and made a new system  image about a month ago. 

 I do have an install from several years ago.  I think I will go back to that one and see if I can get wspr to build.  It could be that newer versions of the rpi package stopped including support for g++ or some dependency.

diyit2, 2020-08-29:
the solution to the problem was to go with a newer OS, 
http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-06-23/
This was for a model B rev 2