JamesP6000/WsprryPi #32: Does not compile with gcc-8.3.0

radio-miskovice, 2019-11-15:
A huge number of errors and warnings during compile:

- mailbox.c:280:52: warning: implicit declaration of function ‘makedev’ [-Wimplici         t-function-declaration]

- In file included from /usr/include/c++/8/vector:64,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/stl_vector.h: In member function ‘void std::__cxx1998::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::__cxx1998::vector<double, std::allocator<double> > >’ changed in GCC 7.1

...and a number of the same errors in different files:

In file included from /usr/include/c++/8/vector:69, from wspr.cpp:40: /usr/include/c++/8bits/vector.tcc: 

/usr/include/c++/8/bits/stl_uninitialized.h:311:2:
/usr/include/c++/8/bits/stl_uninitialized.h:287:5:
/usr/include/c++/8/bits/stl_uninitialized.h:289:37:
/usr/include/c++/8/bits/stl_uninitialized.h:115:5:
/usr/include/c++/8/bits/stl_uninitialized.h:134:15:
/usr/include/c++/8/bits/stl_uninitialized.h:99:9:
... and more

theventurousco, 2020-01-06:
I'm having the same problem. I'm not a very savvy linux user at all. Only recently got into it and my UNIX commands that I learned in college in 92 are rusty at best.

uname -a gave me this info about the OS on my rPi 1. (the first gen one)
Linux raspberrypiI 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux

Has a solve been identified?

damyan-ognyanov, 2020-01-29:
About `makedev` warning : it is due to a change in glibc after version 2.25:

```
* The inclusion of <sys/sysmacros.h> by <sys/types.h> is deprecated.  This
  means that in a future release, the macros “major”, “minor”, and “makedev”
  will only be available from <sys/sysmacros.h>.
```

The fix is to `#include <sys/sysmacros.h>` in `mailbox.c` before `#include <sys/mman.h>`

About the warnings in STL libraries -check [this link](https://stackoverflow.com/questions/48149323)
suggest adding `-Wno-psabi` to `CXXFLAGS` in project's `makefile`

HTH

PE5E, 2020-03-26:
Did a fresh install of the latest Raspbian and got the same problem.
Fix is working as suggested by damyan-ognyanov, so thank you very much.
Can the fix be included in the code?