JamesP6000/WsprryPi #33: WsprryPi on a Pi4 not working

KB8RCO, 2020-02-03:
We did some troubleshooting after trying to run WsprryPi on a Raspberry Pi 4.
Two issues:
1. makefile does not correctly identify the Pi version
   Both the Pi3B and Pi4 identified the same ARMV7l
2. No configuration for PI4 clock changes and base address.

1. makefile I used needs to be edited to properly set PI-VERSION
2. The RPi 4 requires 2 changes in gpioclk
  #define BCM2708_PERI_BASE 0xfe000000
  #define F_PLLD_CLK (750000000.0)
This meant adding RPI4 option in the 'makefile' and changes to use this new setting in both gpioclk.cpp and wspr.cpp.  

I would like to provide updates so others can use the Pi4.

samhanie, 2020-09-14:
Any update to making this software Pi4 compatible?

KB8RCO, 2020-09-21:
There are actually 3 major changes to the Pi4 (vs. older versions) that make the WsprryPi application malfunction with a Pi4:
* BCM2708_PERI_BASE 0xfe000000
*  F_PLLD_CLK (750000000.0)
*  F_XTAL     (54000000.0)

My attempt at fising these issues can be found at under my callsign: https://github.com/KB8RCO/WsprryPi
It adds 6-char grid and GPIO , but must be compiled on the same Pi veriosn it is used and requires the added getbaseaddr executable.  I know 3 people who have used this code and compiled it on multiple Pi version - successfully.

I have another that pulls the base address and other info within the code, so it will execute on any pi, no matter which it was compiled on.  That is still being tested.

KB8RCO, 2020-09-22:
My attempt at fixing these issues can be found at under my callsign: https://github.com/KB8RCO/WsprryPi-1
  It reports which Pi version is detected on start of application.
This was compiled on a Pi4, then run on a Pi3 and Pi Zero to ensure it still worked.
Then compiled on a original Pi (512 MB) and run on a Pi4.

PatKilroy, 2021-01-26:
REFERENCE: "QRPi Software Installation Guide for Different Ham Tx Modes" by Zoltan Doczi, HA7DCD. 

Using the installation sequence given on Page 3 of this guide, under the section, "WSPR-GPIO4," when I hit Enter for the "make" command, I got lines and lines of what appeared to be error codes that I did not understand.  

I request that an experienced user please offer a work around for us newbies, line by line, with typos removed and commands proven for the latest Raspberry Pi OS release, for the Pi 3 and the Pi 4 hardware, if there is any difference in installation sequence between the two models.  I and many others would be very grateful.  Thank you for your support!  

Cheers,

Pat N8PK 
 in FM19

KB8RCO, 2021-01-30:
Sorry for the delayed repsonse, but I had a very hard time finding a readable copy of the referenced document ("WRPi Software Installation Guide for Different Ham Tx Modes" by Zoltan Doczi, HA7DCD).
The newest file on the 'JamesP6000' structure is over 4 years old (APR-2017).  
This is before the Pi3 was well known - justre ad the comments, and certainly before the Pi4.
I made attempts to update to the overall structure to account for versions of the Pi (32-bit at least) through P4. 
Try following the same sintructions but using git clone https://github.com/KB8RCO/WsprryPi-1 as the code repository. 
I know of at least 3 people who have successfully built the code from this repository.If you have issues with that version, email back and I will try and help.

Robert Giuliano
KB8RCO

PatKilroy, 2021-02-01:
Hello Robert,

Thank you very much for your very welcomed attempt!  I'm sorry that I assumed many were familiar with the document as posted on the TAPR dot org site under their products for WSPR, so I see I should have entered a hyperlink to it as https://docs.google.com/document/d/1bLKEDmJvKQegqUwDXEy3V7wQTqupIfyg2LkN0dOuPY4/

... And not only before the Pi3 was well known, but also the current, latest and greatest OS version.  ;-)

I will def go to your repository within the next 48 hours and give it a try.  What could anybody lose!

Thank you again.  Will leave a full report for you and all!

Cheers,

Pat
N8PK-FM19
Sykesville, MD USA

IgrikXD, 2023-10-24:
@KB8RCO 
[image]
В репозитории отсутсвует файл с таким именем. Почему ваша инструкция и реальное содержимое проекта не соттвествует дейтсвительности?

KB8RCO, 2023-10-24:
I do not know how github works, but if you look at the README here:    https://github.com/KB8RCO/WsprryPi-1I had redone the code and removed the install script.
NOTE:  This was never tested on a Raspberry Pi OS AARCH64 Pi (neither Pi3 nor Pi4)     I just tried it and received errors for mbox.virt_addr.  I believe to be OS architecture related. 

Robert Giuliano
KB8RCO

KB8RCO, 2023-10-25:
Verified this code only works fro the 32-bit version of Raspberry Pi OS.the mailbox code causes errors related to addressing.  Unsigned addresses are probably larger, due to the code being compiled fro 64-bit processor.

Robert Giuliano
KB8RCO