JamesP6000/WsprryPi #34: WsprryPi transmit signal quits long before two mins

stiteshl, 2020-03-15:
I am trying to transmit on 20M and running a Pi-3-B+, but this problem also happens on a 3-B or a Zero-W. The program never crashes, but the PTT or transmit signal coming from the Pi stops somewhere between approx. 1-8 secs. The signal never hangs in there for anywhere near 2 mins. With the top-hat removed the software does the same on all three Pi's. This happens whether I am running the code as self-calibration or free-running.  James, is your source code looking for timing from outside the code other than NTP? What do you think the problem is? Thank you, Herbert, WA0TEY

gluck35533, 2020-08-13:
Hello Herbert!

I had the same problem until I disabled 1-wire interface in the rasbbery PI configuration menu.

My wspr beacon is now powered by WsprryPi software and Raspbian Jessie release 2017-07-05. That being said, I had no problems installing and compiling WsprryPi.

Try to follow my advice and disable 1-wire interface in the Rasbbery PI configuration menu.

Best wishes Dmitry, R2DTJ!

stiteshl, 2020-08-15:
Thank you so much Dmitry, your fix has fixed my problem! Are you also into packet radio through the ISS digi-peter?
my direct e-mail is: herbert.stites@hotmail.com