JamesP6000/WsprryPi #37: Compile errors on Raspbian Buster

dslotter, 2020-06-07:
Attempting to build under Raspbian buster (armhf) image from 2020-05-27 results in these errors:

`pi@raspberrypi:~/hamradio/WsprryPi $ make
cc -O2 -march=native -mtune=native -Wall -c mailbox.c
mailbox.c: In function ‘mbox_open’:
mailbox.c:280:52: warning: implicit declaration of function ‘makedev’ [-Wimplicit-function-declaration]
     if(mknod(LOCAL_DEVICE_FILE_NAME, S_IFCHR|0600, makedev(MAJOR_NUM_A, 0)) >= 0 &&
                                                    ^~~~~~~
g++ -O2 -march=native -mtune=native -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI23 mailbox.o wspr.cpp -owspr
wspr.cpp: In function ‘void txon()’:
**wspr.cpp:338:38: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]**
   ACCESS_BUS_ADDR(CM_GP0CTL_BUS) = *((int*)&setupword);
                                     ~^~~~~~~~~~~~~~~~~
wspr.cpp: In function ‘void wspr(const char*, const char*, const char*, unsigned char*)’:
**wspr.cpp:597:10: error: ‘char* strncpy(char*, const char*, size_t)’ specified bound 16 equals destination size [-Werror=stringop-truncation]**
   strncpy(buf, call, 16);
   ~~~~~~~^~~~~~~~~~~~~~~
**wspr.cpp:633:12: error: ‘char* strncpy(char*, const char*, size_t)’ specified bound 4 equals destination size [-Werror=stringop-truncation]**
     strncpy(l, l_pre, 4);
     ~~~~~~~^~~~~~~~~~~~~
**wspr.cpp:693:30: error: ‘j0’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
           j0 = ((k>>j)&1)|(j0<<1);**
                           ~~~^~~~
**wspr.cpp:640:23: error: ‘ng’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   unsigned long n2=(ng<<7)|(p+64+nadd);**
                    ~~~^~~~
`

chrismaltese, 2020-06-14:
Experiencing same, here.  Seeing this was closed, I cannot seem to find the fix with my blind old eyes. Any guidance?

dslotter, 2020-06-14:
@chrismaltese I switched projects to the original (older) version which still builds successfully: https://github.com/DanAnkers/WsprryPi . Hope this helps you.

closed 2020-06-14 by dslotter

chrismaltese, 2020-06-14:
@dslotter Thanks, Dave.  New issue from @DanAnkers build.  I renamed 'makefile1' to 'makefile' for use on my RPi Zero.  Receive the following whilst compiling:

pi@raspberrypi:/WsprryPi $ make
cc -Wall -c mailbox.c
mailbox.c:42:7: error: conflicting types for ‘mapmem’
 void *mapmem(unsigned base, unsigned size)
       ^~~~~~
In file included from mailbox.c:38:
mailbox.h:93:10: note: previous declaration of ‘mapmem’ was here
 uint8_t *mapmem(uint32_t base, uint32_t size);
          ^~~~~~
mailbox.c:70:7: error: conflicting types for ‘unmapmem’
 void *unmapmem(void *addr, unsigned size)
       ^~~~~~~~
In file included from mailbox.c:38:
mailbox.h:99:6: note: previous declaration of ‘unmapmem’ was here
 void unmapmem(uint8_t *addr, uint32_t size);
      ^~~~~~~~
make: *** [makefile:19: mailbox.o] Error 1

I'm stumped.  Have you seen this?

--Chris

chrismaltese, 2020-06-14:
@dslotter Thanks, Dave.  That came together nicely.  Will test with HAT when it arrives.

Much appreciated, 
--Chris