JamesP6000/WsprryPi #41: Small make-stopper with "undefined reference makedev"

MiBatu, 2021-01-25:
The problem is a missing function called makedev, not a program.
makedev() is actually a macro and you need to include the correct header for it.

It is possible  to add the following line in mailbox.c to fix it:
#include <sys/sysmacros.h>

MaxDeepCoder, 2022-02-05:
No doesn't work for me on my rpi 3b

MaxDeepCoder, 2022-02-05:
```
cc -Wall -c mailbox.c
mailbox.c: In function ‘mapmem’:
mailbox.c:62:35: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   62 |         printf("mmap error %d\n", (int)mem);
      |                                   ^
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI23 mailbox.o wspr.cpp -owspr
wspr.cpp: In function ‘void getRealMemPageFromPool(void**, void**)’:
wspr.cpp:265:21: error: cast from ‘unsigned char*’ to ‘unsigned int’ loses precision [-fpermissive]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:265:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:266:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  266 |   *bAddr = (void*)(((unsigned)mbox.bus_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make: *** [makefile:22: wspr] Error 1

```