JamesP6000/WsprryPi #47: Compile error

mucsb, 2021-10-06:
I try to compile your whisper code on RPI. It is a brand new lite image. 
I did 
sudo apt-get update
sudo apt-get upgrade

sudo apt-get install git
git clone https://github.com/JamesP6000/WsprryPi.git

and make gives a nasty error see it beloow. Would you suggest what do I make bad?

Thanks, Béla 73 de HA4BM

pi@whisper:~/WsprryPi $  make
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI1 mailbox.o wspr.cpp -owspr
In file included from /usr/include/c++/8/vector:64,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/stl_vector.h: In member function ‘void std::__cxx1998::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/stl_vector.h:1085:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<double*, std::__cxx1998::vector<double, std::allocator<double> > >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ^~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:69,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/vector.tcc: In member function ‘void std::__cxx1998::vector<_Tp, _Alloc>::_M_realloc_insert(std::__cxx1998::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const double&}; _Tp = double; _Alloc = std::allocator<double>]’:
/usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::__cxx1998::vector<double, std::allocator<double> >::iterator’ {aka ‘__gnu_cxx::__normal_iterator<double*, std::__cxx1998::vector<double, std::allocator<double> > >’} changed in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:63,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/stl_uninitialized.h: In function ‘_ForwardIterator std::__uninitialized_move_if_noexcept_a(_InputIterator, _InputIterator, _ForwardIterator, _Allocator&) [with _InputIterator = double*; _ForwardIterator = double*; _Allocator = std::allocator<double>]’:
/usr/include/c++/8/bits/stl_uninitialized.h:311:2: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
       return std::__uninitialized_copy_a
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
  (_GLIBCXX_MAKE_MOVE_IF_NOEXCEPT_ITERATOR(__first),
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   _GLIBCXX_MAKE_MOVE_IF_NOEXCEPT_ITERATOR(__last), __result, __alloc);
   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h: In function ‘_ForwardIterator std::__uninitialized_copy_a(_InputIterator, _InputIterator, _ForwardIterator, std::allocator<_Tp>&) [with _InputIterator = std::move_iterator<double*>; _ForwardIterator = double*; _Tp = double]’:
/usr/include/c++/8/bits/stl_uninitialized.h:287:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
     __uninitialized_copy_a(_InputIterator __first, _InputIterator __last,
     ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h:287:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
/usr/include/c++/8/bits/stl_uninitialized.h:289:37: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
     { return std::uninitialized_copy(__first, __last, __result); }
              ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h: In function ‘_ForwardIterator std::uninitialized_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = std::move_iterator<double*>; _ForwardIterator = double*]’:
/usr/include/c++/8/bits/stl_uninitialized.h:115:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
     uninitialized_copy(_InputIterator __first, _InputIterator __last,
     ^~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h:115:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
/usr/include/c++/8/bits/stl_uninitialized.h:134:15: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
       return std::__uninitialized_copy<__is_trivial(_ValueType1)
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            && __is_trivial(_ValueType2)
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            && __assignable>::
            ~~~~~~~~~~~~~~~~~~
  __uninit_copy(__first, __last, __result);
  ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h: In static member function ‘static _ForwardIterator std::__uninitialized_copy<true>::__uninit_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = std::move_iterator<double*>; _ForwardIterator = double*]’:
/usr/include/c++/8/bits/stl_uninitialized.h:99:9: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
         __uninit_copy(_InputIterator __first, _InputIterator __last,
         ^~~~~~~~~~~~~
/usr/include/c++/8/bits/stl_uninitialized.h:99:9: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
/usr/include/c++/8/bits/stl_uninitialized.h:101:27: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
         { return std::copy(__first, __last, __result); }
                  ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/8/vector:60,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/stl_algobase.h: In function ‘_OI std::copy(_II, _II, _OI) [with _II = std::move_iterator<double*>; _OI = double*]’:
/usr/include/c++/8/bits/stl_algobase.h:446:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
     copy(_II __first, _II __last, _OI __result)
     ^~~~
/usr/include/c++/8/bits/stl_algobase.h:446:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
In file included from /usr/include/c++/8/bits/stl_algobase.h:67,
                 from /usr/include/c++/8/vector:60,
                 from wspr.cpp:40:
/usr/include/c++/8/bits/stl_iterator.h: In function ‘decltype (std::__miter_base(__it.base())) std::__miter_base(std::move_iterator<_IteratorL>) [with _Iterator = double*]’:
/usr/include/c++/8/bits/stl_iterator.h:1238:5: note: parameter passing for argument of type ‘std::move_iterator<double*>’ changed in GCC 7.1
     __miter_base(move_iterator<_Iterator> __it)
     ^~~~~~~~~~~~
/usr/bin/ld: mailbox.o: in function `mbox_open':
mailbox.c:(.text+0xc54): undefined reference to `makedev'
/usr/bin/ld: mailbox.c:(.text+0xcbc): undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make: *** [makefile:22: wspr] Error 1
pi@whisper:~/WsprryPi $

mbroihier, 2021-10-06:
See issue 41

mucsb, 2021-10-07:
> See issue 41

It works now:
It is possible to add the following line in mailbox.c to fix it:
#include <sys/sysmacros.h>