JamesP6000/WsprryPi #49: Change GPIO From Pin 4 To 17?

pommy159, 2021-12-17:
Hi.
After several years of use, the gpio pin used for WsprryPi (gpio4) has gone faulty, probably due to static.
Is it possible for you to tell me how to change the gpio pin used in WsprryPi from gpio4 to another gpio pin please, for example gpio17
Thank you.

mbroihier, 2021-12-17:
I won't be surprised if simply switching pins doesn't work.  If you fried something, you've likely fried something common to several pins.

Anyway, I believe the trick will be to get GPCLK0 output to a different pin.  If you look in BCM2835 ARM Peripherals (page 102), you'll see that pin 34 has a similar configuration as pin 4 for ALT0.  Pin 34 is in GPIO alternate function select register 3.  I'd first try with this pin.

To direct the output to pin 34, in gpioclk.cpp add a #define GPFSEL3 (0x7E20000C) and change the SETBIT and CLRBIT calls (in txon) to:

SETBIT(GPFSEL3, 14);
CLRBIT(GPFSEL3, 13);
CLRBIT(GPFSEL3, 12);

I have not tried this so no guarantees and double check my addresses and reasoning.  Also, you may need to adjust the drive strength, by switching the "ACCESS" macro call from:

ACCESS(PADS_GPIO_0_27) to
ACCESS(PADS_GPIO_28_45)

where PADS_GPIO_28_45 is defined as 0x7e100030

I haven't found that "drive strength" ever makes a difference in the projects I've done and I find the documentation of this pretty sparse so I tend to not want to monkey with it if it isn't really doing anything I notice.

pommy159, 2021-12-17:
Thankyou very much for taking the time to reply to this. Ill give your recommendation a try later. I have tested all the other gpio pins with a meter whilst toggling them high/low with an application and the others work fine. Thanks for your assistance.

rsoennic, 2022-06-16:
I have this same issue. Did you (@pommy159)  ever get this pin swap to work?

Rich

rsoennic, 2022-06-20:
> 

I worked out how to do this but Mark is correct, it seems all the GPIO common to the "pads" of GPIO4 are damaged. There doesn't seem to be an option to fix it as there are no available clock output pins.