JamesP6000/WsprryPi #50: Compile issue

MaxDeepCoder, 2022-02-05:
Hi my rpi 3b is not working plus i have used issue 41's code and it still doesn't work !

here is the code:

```
cc -Wall -c mailbox.c
mailbox.c: In function ‘mapmem’:
mailbox.c:62:35: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   62 |         printf("mmap error %d\n", (int)mem);
      |                                   ^
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI23 mailbox.o wspr.cpp -owspr
wspr.cpp: In function ‘void getRealMemPageFromPool(void**, void**)’:
wspr.cpp:265:21: error: cast from ‘unsigned char*’ to ‘unsigned int’ loses precision [-fpermissive]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:265:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:266:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  266 |   *bAddr = (void*)(((unsigned)mbox.bus_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make: *** [makefile:22: wspr] Error 1
```

wkraus, 2022-04-12:
Same problem here, tries to add #include <sys/sysmacros.h> but no success.

Exactly same error, working with Rapi 3 B+

Sergey-R5AU, 2023-02-18:
Hi ! i am running on Raspberry Pi 3 Model B Rev 1.2 and have  very close issue within compile

=========================================
pi@time-server:~/WsprryPi $ ls
gpioclk.cpp  gpl-2.0.txt  ISSUES  LICENSE  mailbox.c  mailbox.h  mailbox.o  makefile  README  wspr.cpp
pi@time-server:~/WsprryPi $ make
g++ -D_GLIBCXX_DEBUG -std=c++11 -Wall -Werror -fmax-errors=5  -lm -DRPI23 mailbox.o wspr.cpp -owspr
wspr.cpp: In function ‘void getRealMemPageFromPool(void**, void**)’:
wspr.cpp:265:21: error: cast from ‘unsigned char*’ to ‘unsigned int’ loses precision [-fpermissive]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:265:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  265 |   *vAddr = (void*)(((unsigned)mbox.virt_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
wspr.cpp:266:12: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
  266 |   *bAddr = (void*)(((unsigned)mbox.bus_addr) + offset);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make: *** [makefile:22: wspr] Error 1
pi@time-server:~/WsprryPi $
===============================================

pi@time-server:~/WsprryPi $ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000006aa3b8cc
Model           : Raspberry Pi 3 Model B Rev 1.2
=================================================
pi@time-server:~/WsprryPi $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
==================================
pi@time-server:~/WsprryPi $ uname -m
aarch64
==================================

I have tried on 32bit OS but have similar error retrive , need help

IgrikXD, 2023-10-24:
**Fix:**
Replace wrong code at line **270-271**:
*vAddr = (void*)((uintptr_t)mbox.virt_addr + offset);
*bAddr = (void*)((uintptr_t)mbox.bus_addr + offset);

Sergey-R5AU, 2023-11-01:
Thank you  **IgrikXD** it`s working for me.

Just summary, my setup:  RPI3
--------------------------------------
r5au@wspr:~/WsprryPi $ uname -a
Linux wspr 6.1.29-v8+ #1652 SMP PREEMPT Wed May 24 14:46:55 BST 2023 aarch64 GNU/Linux
--------------------------------------
I did moficitations:
1. https://github.com/JamesP6000/WsprryPi/issues/32#issuecomment-579638900  by **damyan-ognyanov** :
added   **#include <sys/sysmacros.h> in mailbox.c before #include <sys/mman.h>**
and
added  **-Wno-psabi to CXXFLAGS in project's makefile**

2. Updated wspr.cpp with recommendations by **IgrikXD**

and finally:

[image]
[image]

de R5AU

meteoralert, 2024-01-04:
R5AU: Your WSPR transmission as shown in your output only lasted for 81 seconds. WSPR transmissions should have a duration of 110.6 seconds to be decoded properly. I am seeing the same result with my RPI 3B+. Does anyone know what is causing the transmission to end after 81 seconds? Is there a fix to this problem? Thank you!